toppers/hakoniwa #40: Unityのスクリプトのパスの指定方法について

nabenosuke, 2020-06-14:
Webサイトの導入部分でsingle-robot-HackEV.unitypackageを使って、athrillとUnity間で通信するためにEV3ActuatorとEV3SensorにMMAPファイルの絶対パスを指定しようとしているのですが、どこから設定できるのでしょうか。

ytsuji3333, 2020-06-15:
導入手順の「導入の流れ」‐「6.Unityのインストール・設定」‐「初回に行う設定」にて
「MMAP版を使用する場合はこちら」のリンクを開きます．

UDP用Unity設定（実際はMMAP用Unity設定です．誤記であるため後ほど修正いたします．）
に記載の通り，`Scripting Define Symbols` に`VDEV_IO_MMAP`と設定することで
EV3ActuatorとEV3Sensorに「Filepath」というパラメータが表示されるようになります．
そちらに，MMAPファイルの絶対パスを指定してください．

nabenosuke, 2020-06-15:
`Scripting Define Symbols`に`VDEV_IO_MMAP`を入力するとEV3ActuatorとEV3SensorのStime_offやFilepathを設定できるウィンドウが出てくるという認識で合っているのでしょうか.

ytsuji3333, 2020-06-15:
はい，ご認識の通りです．

takasehideki, 2020-06-16:
@nabenosuke さん
当初の内容が解決されたようでしたら，このIssueはCloseとしてください．
また異なる質問が出てきたら，改めてIssueを立ててください．

nabenosuke, 2020-06-17:
Filepathを設定するウィンドウが出てこないので一から導入をやり直そうとしたところ別のところで詰まってしまったので、一度別のissueでその問題を解決してから取り組みなおそうと思います。

ytsuji3333, 2020-06-17:
こちらでも，再度設定を変更できるか確認してみました．
あまり意識していなかったのですが，`VDEV_IO_MMAP` と設定したあと，カーソルを一度外してみたり，
`Other Settings` タブを折りたたむなどしてみてください．
また，設定が反映されるまで，少し時間がかかりますので少々お待ちいただければと思います 🙇
 
[image]

nabenosuke, 2020-06-18:
実際の画面操作を見せていただきとても分かりやすかったです。
設定後、HietatchyからRobotを選択し、InspectorのAdd ComponentからEV3ActuatorとEV3Sensorを追加したところ、そちらの画面通りにFilepathを設定できるようになりました。
このような手順で良いのでしょうか？

ytsuji3333, 2020-06-18:
> Filepathを設定できるようになりました。

であるならば，問題ないと思います．

nabenosuke, 2020-06-18:
わかりました。
パスの指定方法についてもう一つわからないことがあるのですが、MMAPファイルの絶対パスとは、例えば次のシミュレータ手順で`workspace`の中の`line_trace`を使う場合は、`line_trace`の中のathrill_mmap.binとunity_mmap.binを指定すればよいのでしょうか

ytsuji3333, 2020-06-18:
はい，ご認識の通りです．

nabenosuke, 2020-06-27:
できました。ありがとうございました

closed 2020-06-27 by nabenosuke