toppers/hakoniwa #45: OGPとTwitterCardを設定する

takasehideki, 2020-06-16:
ここが参考になりそう
https://exlair.net/hugo-internal-templates-ogp-and-twitter-card/

mitsut, 2020-06-17:
関連しそうなページ
https://discourse.gohugo.io/t/optimizing-seo-for-blog/19873/17
https://miyahara.hikaru.dev/posts/20200319/

mitsut, 2020-07-04:
Facebookでのデバッグ表示のページ（[シェアデバッガ](https://developers.facebook.com/tools/debug/)）
Twitterでのデバッグ表示のページ（[Card Validator](https://cards-dev.twitter.com/validator)）

mitsut, 2020-07-04:
似たようなissueがあった
https://github.com/keitakawamoto/orangebomb.org/issues/117

Kubeのテンプレートではすでに、OGPは対応済みになっている。
kube/layouts/partials/meta/ogimage.html

config.tomlにimagesの情報がないので、フロントマターでアイキャッチの指定がないとなにも出力されないという状況になっている。

とりあえず、hakoniwaロゴを指定するようにする

takasehideki, 2020-07-09:
すみませんここに貼ったつもりだったんですが，ボケてたようです．
ひとまずOGP向けの画像サイズが小さいみたいですね．
https://github.com/toppers/hakoniwa/pull/54#issuecomment-655841812

mitsut, 2020-07-09:
Facebookに関する仕様（リンクシェアの画像）
https://developers.facebook.com/docs/sharing/webmasters/images

mitsut, 2020-07-09:
Twitter Cardに関するドキュメント
https://developer.twitter.com/en/docs/tweets/optimize-with-cards/overview/abouts-cards

takasehideki, 2020-07-10:
@mitsut  #54 と #57 で対応完了されたことを確認しました．あざまっした！！

mitsut, 2020-07-10:
Hugoのinternal templateの振る舞いがどのようになっているのか

Hugoのinternal template に関する記述
https://gohugo.io/templates/internal/

Githubのソースコード
https://github.com/gohugoio/hugo/blob/release-0.73.0/tpl/tplimpl/embedded/templates/opengraph.html
https://github.com/gohugoio/hugo/blob/release-0.73.0/tpl/tplimpl/embedded/templates/twitter_cards.html

Internal templateを使った記事
https://exlair.net/hugo-internal-templates-ogp-and-twitter-card/

mitsut, 2020-07-10:
記事のイメージを使う場合の手順についてはREADME-webに書くことにして本件は閉じます

closed 2020-07-10 by mitsut