toppers/hakoniwa #73: サンプルアプリケーションの使い方について

nabenosuke, 2020-08-14:
導入手順、使用手順に沿ってV850版で`single-robot-HackEV(v2.0)`を使って`line_trace`と`color_sensor`を動かそうとしたところUDP版の`color_sensor`の場合だけ動き、MMAP版や`line_trace`だとロボットが動かないのですが、何が違うのでしょうか。

なおWindowsから使っており、Unity以外はWSL上に配置していて、MMAP版を使う際はMMAPファイルのパスを`\\wsl$\Ubuntu-18.04\home\username\ev3rt-athrill-v850e2m\sdk\workspace\color_sensor(line_trace)\unity_mmap.bin(athrill_mmap.bin)`と指定しています。

ytsuji3333, 2020-08-17:
■UDP版の`color_sensor`だけ動き，MMAP版の`color_sensor`が動作しない件について
　UDP版とMMAP版の違いとして，athrill実行時の引数に指定するファイルが異なる
　という点があります．
　UDP版の場合
　```$ athrill2 -c1 -t -1 -m memory.txt -d device_config.txt ../asp```
　MMAP版の場合
　```$ athrill2 -c1 -t -1 -m memory_mmap.txt -d device_config_mmap.txt ../asp```

　また，MMAP版ではUnityのシミュレータ起動時にmmapファイルの初期化を行っているため，
　Unityのシミュレータの起動→athrillの起動の順で実行する必要があります．

■`line_trace`でロボットが動かない件について
　`color_sensor`との違いは特にありません，MMAP版ではファイルパスが
　異なるぐらいだと思います．

　UDP版の場合，`color_senosr`と異なるポート番号を使用しているのであれば，
　`color_sensor`で動作していたポート番号を使用してみてください．
　MMAP版の場合，再度MMAPファイルのパスが正しいかご確認ください．
　ファイルパスが正しくないなどの理由で，MMAPファイルが読み込めなかったとき
　Unity側の画面下部の`Console`タブ内で，MMAPファイルが読み込めなかった旨の
　エラーが出力されます．

nabenosuke, 2020-08-30:
ありがとうございます。
Unityのシミュレータを起動した後にUDP版、MMAP版に対応したファイルを引数としてathrillを実行しましたが、動作しませんでした。

`line_trace`を動かす際ポート番号を確認しましたが、`color_sensor`と同じポート番号であり、エラーも出力されませんでした。

また、環境が6月ごろに導入した時のままでしたので、それぞれgit pullで新しくしたところ、UDP版の`color_sensor`も動かなくなりました。
今まで動かなかった他の3つパターンも、`#### motor control start`までは出力されていた覚えがあるのですが、それも出力されなくなりました。

ytsuji3333, 2020-08-30:
原因の切り分けのために，導入・使用手順のどの段階まで進んでいるのか確認させてください．
- [ ] athrillのビルドに成功(ビルド後，ビルドを行ったフォルダに`timer32.o`が作成されるかご確認ください)
- [ ] 制御アプリのビルドに成功(ビルド後，`asp`という実行ファイルがビルドを行った時間で作成されているかご確認ください)
- [ ] Unity側でのシミュレータ起動(シミュレータ起動時，エラーが発生していないかご確認ください)
- [ ] athrillの実行(エラーなく実行されるか，`#### motor control start`のメッセージが出力されるかご確認ください)

nabenosuke, 2020-08-31:
申し訳ありません。環境を見直す過程で.bashrcのPATH設定でタイプミスを起こしていました。
また、`line_trace`のUDP版が動作しない件は、device_config.txtを弄ってしまっていたことが原因でした。
これらを修正したところ、athrillのビルドからathrillの実行まで成功していることを確認しました。

しかし、MMAP版はどちらも動作しませんでした。
Unityの置くべき場所を把握できていないのですが、Windowsでの導入手順ではWindows上のUnityとWSL上のathrill間でMMAP通信を行っているのでしょうか。

ytsuji3333, 2020-08-31:
> Unityの置くべき場所を把握できていないのですが、Windowsでの導入手順ではWindows上のUnityとWSL上のathrill間でMMAP通信を行っているのでしょうか。

はい，その通りだと私は認識しています．
一度，athrillなどの単体ロボット向けシミュレータに関わるリポジトリをWIndowsのフォルダ(例えばCドライブの任意のフォルダ)などに場所を変えて試していただくことは可能でしょうか？

nabenosuke, 2020-09-01:
リポジトリをCドライブ上のフォルダに移したところ、どちらも動作させることができました。ありがとうございました。

`color_sensor`でロボットがコースを一周した後コースを外れて落下するのですが、これは仕様でしょうか？

tmori, 2020-09-01:
> リポジトリをCドライブ上のフォルダに移したところ、どちらも動作させることができました。ありがとうございました。
> 
> `color_sensor`でロボットがコースを一周した後コースを外れて落下するのですが、これは仕様でしょうか？

仕様ではありません．
ライントレース制御パラメータのチューニングがうまく行ってないのだと思います．
単体ロボットシミュレータとしては，あくまでもサンプルプログラムの提供(動くところまで)なので，完全な制御ができているプログラムは可能ならやりたいですが，必須ではないという認識でおります．

nabenosuke, 2020-09-12:
把握いたしました。
ありがとうございました。

closed 2020-09-12 by nabenosuke