toppers/hakoniwa #84: ETロボコン2020のシミュレーション環境の制御系設計法の研究利用について

ToshitakaOki, 2020-10-21:
お世話になります。
　　　福山大学の沖と申します。

ご相談があります。
現在、ETロボコン2020のシミュレーション環境でライントレースロボットの制御器の設計と検証の研究を行いたいと考えています。
そこで、ETロボコンのホームページから辿れる v.0.992 を使わせて頂きたい次第です。

理由は、シミュレーション環境なので、ベンチマークテストの様に利用でき、第３者が制御系設計手法と制御性能の検証を行うことができることを期待しています。

そのため、制御系設計法の研究材料としての利用、および、学会等での発表をさせて頂きたいと考えています。

一度ETロボコン事務局にご相談させていただいたところ、箱庭様にお問い合わせするよう指示をいただきました。

このご指示に先立ち、次のご意見もいただいております。
****
ETロボコンのHPに公開されている「評価版シミュレータ」は
評価版シミュレータのご利用は、以下の用途に限らせていただきます。
https://github.com/ETrobocon/etrobo/
---
・ETロボコンへの参加のため
・ETロボコンへの参加を検討するため
・ETロボコンの広報
・ETロボコンに参加、協力していることを広報するため
---
> 基本的には上記の範囲内で使用していただいておりますが、
内容によっては検討させていただきます。
制御の研究で使用したいとのことですが、
具体的にはどのような研究内容で使用されますか。
****

ご検討のほど、宜しくお願い申し上げます。

takasehideki, 2020-10-22:
@ToshitakaOki 沖先生，まずはお問合せいただきまして，誠にありがとうございます．そして制御系技術の研究題材として箱庭の活用に興味を持っていただき，私たちとても光栄ですし非常に興味があります！

---
まずは，ETロボコン実行委員会で開発・公開されいているシミュレータと，私たちのTOPPERSプロジェクト箱庭WGで開発しているものは，似て非なるものであることを説明させていただきます．

当方では，IoT/自動運転時代の新たな仮想シミュレーション環境である「箱庭」の研究開発に取り組んでいます．コンセプトや狙いを掲げており，その実証のためにいくつかのプロトタイプモデルの実装・実現を推進しています．そのうちのひとつである「単体ロボット向けシミュレータ」は，物理シミュレータとマイコンシミュレータの連携技術の確立を目指して，ETロボコンを題材として実装を進めてきていました．
ETロボコン実行委員会では，箱庭WGの活動に興味を持っていただき，また昨今の時代背景もあり，シミュレータ環境の開発に取り組んでこられてきました．当WGの開発成果のひとつであるAthrill（組込みマイコン向けISAシミュレータ）を採用いただき，可視化・物理演算に該当するUnity側は独自に開発を進められています．

つまり，Athrillは共通利用していまして，またAthrillとUnityの連携技術に関しては情報交換を進めてほぼ共同開発のかたちで実現しましたが，Unity部分およびこれに付随する機能は大きく異なります．

ETロボコンで公開されている「評価版シミュレータ」および大会で使用されている環境は，お問い合わせ回答内容の通り，ETロボコンの参加に限っての利用となっていると伺っています．「評価版シミュレータ」がご提案の研究題材として利用可能であるかは，引き続きそちらにお問合せいただければと思います（私たちには関与できないところとなります）

---
我々の箱庭環境に関しては，ぜひともご提案の研究題材の利用として検討いただきたく考えております．規約としては[TOPPERSライセンス](https://www.toppers.jp/license.html)を掲げておりますが，自由な二次利用を妨げるものではありません．

ETロボコンのシミュレータは全国規模の大会に利用できるほど完成度の高いものになっています．ただし，大会での利用に特化しているものと認識しています．
一方で，我々の開発している箱庭は，カスタマイズ性に利点があるものと考えています．ETロボコン大会のコースおよび競技ルールにとらわれず，ご自身の要望に合わせた環境や設定を自由に構築することができます．現在は，オンライン教育の現場での採用が増えてきています．教育内容や実習課題に合わせたコースやルールの環境整備を行い，学生各自のPCでリモート開発実習を進められています．このあたりに関しては，先日開催されました[Unity道場 ロボティクススペシャル](https://youtu.be/kmnB7qbL-hQ?t=1104)にて解説しておりますので，ご視聴いただけましたらと思います．
このように，研究題材としてのカスタマイズについて，技術的な協力や相談に対応可能です．また，ベンチマークセットとして二次公開されることも大歓迎です．

よろしければより詳細な説明や，沖先生のご要望や利用想定などをお伺いする場を設定できればと考えております．
いかがでしょうか？ ご検討のほどよろしくお願いいたします．

ToshitakaOki, 2020-10-28:
箱庭様

ご回答ありがとうございます。
2次利用の許諾、ありがとうございます。

ETロボコンのシミュレータの利用を考えたのは、開発環境構築が簡単なことが要因で、ベンチマークとして他の人が試す時に敷居が低いと考えたからです。そこで、「評価版シミュレータ」なら別目的でも利用できるかと考えた次第です。ETロボコンの事務局様からは、先の説明とともに、使用の可否を箱庭様で検討いただくようにとのことだったのですが、今では箱庭環境を使用させていただくのがいいように思っています。

> よろしければより詳細な説明や，沖先生のご要望や利用想定などをお伺いする場を設定できればと考えております．
> いかがでしょうか？ ご検討のほどよろしくお願いいたします．

私としましても、願ってもない有難いことです。
ただ現状で、「ライントレースロボットの実環境に近いシミュレーションが容易にできる」ということ以外、明確な仕様があるわけではありません。
まず、これを機会に箱庭環境のことを勉強してみようと思います。

今後ともよろしくお願い申し上げます。

takasehideki, 2020-11-10:
@ToshitakaOki 沖先生，返事が遅くなりまして，誠に申し訳ありません．
ぜひとも箱庭環境を試用いただければと思います．使ってみていろいろアイデアなど湧いてくるようでしたら私たちも嬉しいです．使用方法など分からないところや使い勝手のよくないところがありましたら，遠慮なくお知らせください，全力でサポートして応えていきます．

打合せや説明の場の設定をご希望でしたら，下記までご連絡いただければと思います．どうぞよろしくお願いいたします．
 takase _at_ i。kyoto-u。ac.jp　※ _at_ -> @  / 。-> .

ToshitakaOki, 2020-11-14:
箱庭様

ご連絡とご提案、ありがとうございます。

検討が進みましたら、改めてご相談させていただくこともあるかと存じます。
その時は、どうぞよろしくお願い申し上げます。

takasehideki, 2021-01-26:
この件，ひとまず Close とさせていただきます．
またなにかありましたら Reopen いただくかメールにて問合せいただければと思います．

closed 2021-01-26 by takasehideki