toppers/hakoniwa #87: 「制御アプリのビルド」でエラー

ToshitakaOki, 2020-11-14:
お世話になります。

「プロトタイプモデル・単体ロボット向け　ARM版」を次の環境で構築を試みています。
・VMWare Player 15 + Ubuntu 20.04-2
・single-robot-HackEV(v2.0)：ARM版

ホームページ「[単体ロボット向け](https://toppers.github.io/hakoniwa/prototypes/single-robot/) 」に従って進めているのですが、「[使用方法(TOP)](https://toppers.github.io/hakoniwa/single-robot-usage/single-robot-usage-index/)」で、「[single-robot-HackEV(v2.0)を使用している場合：ARM版](https://toppers.github.io/hakoniwa/single-robot-usage/01_usage_arm_v2.0/)  」において、「EV3ロボット制御プログラムのビルド」の
　make clean;make
をおこなうと、次のエラーで停止します。（長いので最後のあたりだけ示します。）
```
arm-none-eabi-ar -rcs libkernel.a  target_support.o gic_support.o core_support.o  target_config.o tlsf.o pl310.o sf_boot.o chip_config.o chip_timer.o gic.o core_config.o arm.o startup.o task.o wait.o time_event.o task_manage.o task_refer.o task_sync.o task_except.o semaphore.o eventflag.o dataqueue.o pridataq.o mailbox.o mempfix.o time_manage.o cyclic.o alarm.o sys_manage.o interrupt.o exception.o 
arm-none-eabi-ranlib libkernel.a
make: *** 'banner.o' に必要なターゲット '/usr/lib/arm-none-eabi/newlib/fpu/libc.a' を make するルールがありません.  中止.
```
対策を教えていただければ幸いです。
宜しくお願い申し上げます。

tmori, 2020-11-14:
「プロトタイプモデル・単体ロボット向けシミュレータ」は，v850版が安定しておりますので，そちらをご利用いただいたほうがトラブル等発生する可能性は低いと思います．

もし，ARM版での環境が前提と言うことでしたら，少々お時間はかかると思いますが，問題対応させて頂きます．
※１週間以内には対応できるかと思います．

ToshitakaOki, 2020-11-17:
ご連絡ありがとうございます。
ARM版の修正ありがとうございます。
追って試してみたいと思います。

まず、v850で試しています。
HPに従ってインストールし（たつもり）、「単体ロボット向けシミュレータ使用手順(V850版)」に従ってlinetraceのビルドを行うとし、次が発生しました。

```
$ make img=line_trace

rm -rf /home/oki/LineTrace_v850/ev3rt-athrill-v850e2m/sdk/workspace/.././OBJ/
cd /home/oki/LineTrace_v850/ev3rt-athrill-v850e2m/sdk/workspace/.././OBJ/ && \
ruby /home/oki/LineTrace_v850/ev3rt-athrill-v850e2m/sdk/workspace/../../configure.rb  -T v850_gcc -A app \
	-a ../workspace/line_trace \
	-D ../.. \
	-l c \
	-m ../common/Makefile.img \
	-o "" \
	-t \
	-d ./dep \
	-C ../common/athrill_common.cdl \
	-U "" && \
make clean
Traceback (most recent call last):
	2: from /home/oki/LineTrace_v850/ev3rt-athrill-v850e2m/sdk/workspace/../../configure.rb:46:in `<main>'
	1: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:92:in `require'
/usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:92:in `require': cannot load such file -- shell (LoadError)
make: *** [../common/Makefile.workspace:147: prepare-obj-folder] エラー 1
```

お手数をおかけしますが、対処方法を教えていただければ幸いです。

tmori, 2020-11-17:
お問い合わせいただきありがとうございます．

rubyのエラーの件ですね．以下を実行することで対処できます．

```
$ sudo gem install shell
```

また，v850には，インストーラがございます．
下記に手順書を添付しますので，ご参考にして頂けますと幸いです．

[installerを使用した導入手順.pdf](https://github.com/toppers/hakoniwa/files/5550698/installer.pdf)

ToshitakaOki, 2020-11-17:
早速のご連絡ありがとうございます。
`$ sudo gem install shell
`
でビルドができ、シミュレータも動作しました。
ありがとうございました。

ToshitakaOki, 2020-11-17:
ARM版です。
Makefile、device_config.txt、memory.txt を更新し、
`make ATHRILL_BUILD_TARGET=ubuntu20
`でビルドできることを確認しました。
ありがとうございます。

続いて、動作確認をしようとしたところ、躓いてしまいました。

方法1：ev3rt-athrill-ARMv7-A/sdk/OBJ1.1　で、athrill2 -c1 -t -1 -m memory.txt -d device_config.txt ../asp　を実行。
```
oki@ubuntu:~/LineTrace/ev3rt-athrill-ARMv7-A/sdk/OBJ1.1$ athrill2 -c1 -t -1 -m memory.txt -d device_config.txt ../asp
ERROR can not open ../asp
```
方法2：ev3rt-athrill-ARMv7-A/sdk/OBJ1.1　で、athrill2 -c1 -t -1 -m memory.txt -d device_config.txt ./asp　を実行。（../aspを./aspに変更）
```
oki@ubuntu:~/LineTrace/ev3rt-athrill-ARMv7-A/sdk/OBJ1.1$ athrill2 -c1 -t -1 -m memory.txt -d device_config.txt ./asp
core id num=1
ROM : START=0x18000000 SIZE=16
ROM : START=0x18004000 SIZE=1024
RAM : START=0x18204000 SIZE=1024
RAM : START=0x20000000 SIZE=10240
RAM : START=0x20a00000 SIZE=10240
oki@ubuntu:~/LineTrace/ev3rt-athrill-ARMv7-A/sdk/OBJ1.1$ 
```
となり、終了してしまう。

度々お手数をおかけしますが、対象方法を教えて頂ければ幸いです。

ToshitakaOki, 2020-11-17:
ご連絡、ありがとうございます。
次となりました。
動作できないようです。

oki@ubuntu:~/LineTrace/ev3rt-athrill-ARMv7-A/sdk/OBJ1.1$ athrill2 -c1 -t -1 -m memory.txt -d device_config.txt asp
core id num=1
ROM : START=0x18000000 SIZE=16
ROM : START=0x18004000 SIZE=1024
RAM : START=0x18204000 SIZE=1024
RAM : START=0x20000000 SIZE=10240
RAM : START=0x20a00000 SIZE=10240
oki@ubuntu:~/LineTrace/ev3rt-athrill-ARMv7-A/sdk/OBJ1.1$

tmori, 2020-11-17:
方法２が正しいですね．

athrill2ですが，V850版のathrill2が使われている可能性があります．
以下で，athrill2のバージョン/ターゲットをご確認いただけます．
```
$ athrill2
Athrill is licensed under the TOPPERS License Agreement (http://www.toppers.jp/en/license.html).
ARCH:V850e2m (VERSION CORE:1.1.1 TARGET:1.0.0)
```

ToshitakaOki, 2020-11-17:
ありがとうございます。
ご指摘の通りでした。

```
oki@ubuntu:~/LineTrace/ev3rt-athrill-ARMv7-A/sdk/OBJ1.1$ athrill2
Athrill is licensed under the TOPPERS License Agreement (http://www.toppers.jp/en/license.html).
ARCH:V850e2m (VERSION CORE:1.1.1 TARGET:1.0.0)

Usage:athrill -c<core num> -m <memory config file> [OPTION]... <load_file>
 -c                             : set core num. if -c is not set, core num = 2.
 -i                             : execute on the interaction mode. if -i is not set, execute on the background mode.
 -r                             : execute on the remote mode. this option is valid on the interaction mode.
 -t<timeout>                    : set program end time using <timeout> clocks. this option is valid on the background mode.
 -m<memory config file>         : set athrill memory configuration. rom, ram region is configured on your system.
 -d<device config file>         : set device parameter.
oki@ubuntu:~/LineTrace/ev3rt-athrill-ARMv7-A/sdk/OBJ1.1$ 

```

tmori, 2020-11-17:
でしたら，V850版のathrill2のパスをARM版のathrill2に変更いただければ動くと思いますね．

ToshitakaOki, 2020-11-17:
ありがとうございます。
.bashrcで次の様に設定したところ、シミュレータが動作しました。

大変助かりました。
ありがとうございました。

```
#for ARM
export PATH=~/LineTrace/athrill/bin/linux:${PATH}

#for v850
#export PATH=~/LineTrace_v850/athrill/bin/linux:${PATH}
#export PATH=/usr/local/athrill-gcc/bin/:${PATH}
#export LD_LIBRARY_PATH=/usr/local/athrill-gcc:/usr/local/athrill-gcc/lib:${LD_LIBRARY_PATH}
```

takasehideki, 2021-01-26:
動作したとのことですので， Close とさせていただきます．
またなにかありましたら Reopen されるか，新しく Issue を作成いただければと思います．

closed 2021-01-26 by takasehideki

ToshitakaOki, 2021-01-26:
色々お世話になりました。
このところ手が回らなかったのですが、また試してみようと思います。

ありがとうございました。