toppers/hakoniwa #134: コンテンツの英語化

mitsut, 2022-05-14:
- 英語ページは `en/` ポリシーで作る
    - [ ] やりかたをさがす(@mitsut)

- 英語化の対象ページ
    - [ ] TOPの「お知らせ」 `_index.md`
        - 「トピックス・イベント案内」「更新情報」は空で良い．
    - [ ] 「箱庭とは」の全部．図は元データを探す． `docs/`
    - [ ] 「プロトタイプモデル」 `prototypes/`
        - `single_robot`, `ros2sim` へのランディングページ（工数的に余裕があれば multiplay も）
        - [ ] まず日本語の整理から (@s-hosoai)
    - [ ] 「リポジトリ」 repositories.md
    - [ ] 「SNS・問合せ先」 contact.md
   
- リポジトリの README を英語化
    - [ ] https://github.com/toppers/hakoniwa-single_robot
        - appendix/version.md
    - [ ] https://github.com/toppers/hakoniwa-ros2sim
        - appendix/version.md appendix/native.md
    - [ ] 余力があれば https://github.com/toppers/hakoniwa-ros-multiplay
        - [ ] Qiita記事からの移植がまず必要(@tmori )

takasehideki, 2022-05-14:
リポジトリの README 英語化は，対象の md ファイルを明記しました．
まだ今後に増える可能性はありますが，[余力があれば] のラベルをちゃんと付けて区分けします．

takasehideki, 2022-06-15:
リポジトリの README 英語化は，それぞれのリポジトリの Issue に移動させました

https://github.com/toppers/hakoniwa-single_robot/issues/16
https://github.com/toppers/hakoniwa-ros2sim/issues/35
https://github.com/toppers/hakoniwa-ros-multiplay/issues/5

takasehideki, 2022-06-15:
これから各ページの軽いチェックと図の英語化を進めていきます 🚀

takasehideki, 2023-04-18:
already closed by #143

PR #143 (2022-06-15): publish English version of web site
fix #134
closed 2023-04-18 by takasehideki